byuccl/spydrnet-tmr #9: Doing TMR on an EBLIF netlist fails

jacobdbrown4, 2022-02-11:
I tried running an eblif netlist through our TMR tools and it failed. There were several reasons for this:

1. **Finding voter insertion points** 
        find_after_ff_voter_points() calls load_primitive_info() to know what primitives to look for. load_primitive_info() tries to look in a primitives_library in the netlist, but eblif doesn't have libraries, so this fails.
**Proposed fix:** if one does not want to use load_primitive_info() to get the flip flop primitive names, they can pass in list of primitive names to look for. This is what we did before but, in an effort to make things more automated, this was changed. It might be good to have both options. This could solve both the eblif problem and the problem that only XILINX primitive info is supported. Another idea is to have it so if a primitives library is not found, just use the first found library in the netlist (this would work for eblif because the parser right now just puts everything into a single library)

2. **Inserting voters**
        After finding voter insertion points manually, inserting the voters failed too. Our function that inserts voters looks for an item in the metadata called "EDIF.identifier" several times. However, this is a property that only EDIF netlists will have. This causes voter insertion to fail for both Verilog and eblif netlists.

3. **Voters**
        Even if insertion succeeds, although my eblif netlist was synthesized for XILINX, the XILINXTMRVoter won't work because it is meant for an EDIF netlist.

I'll work on finding good solutions to all of these problems.

wirthlin, 2022-02-14:
This is good feedback. We need to come up with some ways to make the tools more general for both non-EDIF and non-Xilinx devices. I would like to start a discussion on how the tools could change to support both cases.

1. It seems that we need to have some sort of library no matter what netlist format or part we are using. Since you are using a Xilinx device, can't we import a default Xilinx library? I think it would be good to start thinking about how we will support other families. It seems that the concept of a library is independent of eblif or any netlist format. A primitive library should be something built into SpyDrNet.

2. All netlists should have a notion of an "identifier". We definitely don't want to have an algorithm that requires "EDIF.identifir". Is there not a built in identfier mechanism with SpyDrNet? This seems like we will need to change the TMR algorithsm.

3. Again, we need to make this more general and support SpyDrNet netlists, not EDIF netlists.

jacobdbrown4, 2022-02-25:
I was able to fix problem #1 with only 2 lines of code. Now the load_primitive_info can find the primitives in the Eblif netlist.

jacobdbrown4, 2022-03-04:
I was able to fix problem 2 with some minor edits to the *organ_insertion.py* file.

jacobdbrown4, 2022-03-04:
I changed how the XilinxTMRVoter works to fix problem 3. I added a parameter to XilinxTMRVoter() that specifies what netlist format to use. The default will give a voter that will work in an EDIF netlist, so all previous scripts will continue to work. To specify a voter that will work in a BLIF/EBLIF netlist, one can do XilinxTMRVoter(netlist_type="BLIF").

closed 2022-04-13 by jacobdbrown4