byuccl/spydrnet-tmr #11: Finding Voters After Symbiflow Renames Primitives

jacobdbrown4, 2022-04-13:
Symbiflow renames some primitives during symbiflow_synth. For example, an FDRE is renamed to be FDRE_ZINI. This causes spydrnet's function to find voter insertion points to come up empty.

wirthlin, 2022-04-13:
Do you know if the tool is creating a wrapper around the FDRE or is it indeed a leaf cell? [Dr. Nelson](https://github.com/nelsobe) thinks there is a wrapper. @nelsobe

nelsobe, 2022-04-13:
ZINI appears all through Symbiflow.   Symbiflow usually uses ZINI to refer to initialization.  So, it might be wrapper for a flip flop that is initialized to a 1 or to a 0 or something like that.  I would do some digging into that.

Also, one of the issues I have with Symbiflow is they rename lots of things which makes it hard to inter-operate with Vivado.  For example - they name some BEL properties differently than Vivado names them...   It sounds ike this is another case of that maybe?

jacobdbrown4, 2022-04-14:
In the netlist it shows up as a leaf cell.

It probably is another case of renaming something to possibly carry information.

In ~/opt/symbiflow/xc7/install/share/symbiflow/techmaps there are files with verilog modules. Here is one for FDRE:

    module FDRE (output reg Q, input C, CE, D, R);
    
    parameter [0:0] INIT = 1'b0;
    
    wire CE_SIG;
    wire SR_SIG;
    
    CESR_MUX cesr_mux(
        .CE(CE),
        .SR(R),
        .CE_OUT(CE_SIG),
        .SR_OUT(SR_SIG)
    );
    
    FDRE_ZINI #(.ZINI(!|INIT), .IS_C_INVERTED(|0))
      _TECHMAP_REPLACE_ (.D(D), .Q(Q), .C(C), .CE(CE_SIG), .R(SR_SIG));
    
    endmodule

FDRE_ZINI shows up at the bottom.

jacobdbrown4, 2022-04-14:
A lot of things are renamed. For example:
OBUF --> OBUFT_VPR
IBUF --> IBUF_VPR
INV --> T_INV

jacobdbrown4, 2022-04-26:
This page may be helpful to us: [**Technology mapping for VPR**](https://f4pga.readthedocs.io/en/latest/flows/f4pga.html#technology-mapping-in-f4pga-toolchain)